AFCProject/AFC-Klipper-Add-On #16: Endless Retraction on reboot.

GavinHanson, 2024-09-12:
I keep getting the same issue where if the printer thinks that there is a tool loaded i.e. leg1 is set to "tool_loaded": true and for whatever reason the filament is NOT loaded it will endlessly try and move the leg as part of prep.

It should check if the toolhead, hub & lane switches are active before it tries to infinitely recover the filament position, and maybe error out the leg if needs be.

GavinHanson, 2024-09-12:
On closer inspection, it looks like i had a stringer in my hub, so HUB_NOT_CLEAR was being shown, and it then systematically ejected all 5 lanes, then espooled them back so they all unraveled 🤦

MG-longshot, 2024-09-13:
Yes if the spools are not stopping that means you have a trigger error cause it's trying to move until the are clear

GavinHanson, 2024-09-13:
> Yes if the spools are not stopping that means you have a trigger error cause it's trying to move until the are clear

Ok, perhaps we need a max-short-move-tries variables and maybe a way to 'ram' the hub using a leg, i assume in my case its a stringer causing it so forcing a leg in there would clear the whisp.

weemantella, 2024-09-15:
With the recent error adding this issue is mostly resolved. This does not take care of the hub if it is still triggered but will not infinitely retract the filament spools.

Additionally, the ```PREP``` command is a great command to run at any time to help BT determine where its filament is or where errors might be.

closed 2025-02-23 by MG-longshot