AFCProject/AFC-Klipper-Add-On #128: AFC shutdown. Timer too close

53Aries, 2024-11-24:
If a lane is loaded on boot, upon performing a toolhead load (T*) as soon as the HUB sensor is triggered klipper will shutdown. I've found the only way to avoid this is to have all lanes empty during boot. If I perform a firmware restart with lanes loaded issue will persist. Lanes must be empty and klipper service restarted to avoid. 
[image]
[image]

[klippy (59).log](https://github.com/user-attachments/files/17892525/klippy.59.log)

jimmyjon711, 2024-11-24:
Have you already updated your trsync values in klipper/klippy/mcu.py file to the following values? Make sure you restart your printer after making this change.

```
TRSYNC_TIMEOUT = 0.05
TRSYNC_SINGLE_MCU_TIMEOUT = 0.50
```

53Aries, 2024-11-24:
> Have you already updated your trsync values in klipper/klippy/mcu.py file to the following values? Make sure you restart your printer after making this change.
> 
> ```
> TRSYNC_TIMEOUT = 0.05
> TRSYNC_SINGLE_MCU_TIMEOUT = 0.50
> ```

I changed these values and had momentary success but on the third toolhead load the issue returned.

closed 2025-02-23 by MG-longshot