AFCProject/AFC-Klipper-Add-On #209: feature request: SET_LANE_LOADED command

thinkl33t, 2024-12-27:
Currently, if the printer software crashes part way through a print, or there is a hardware fault you can be left in a position where the only way to recover your AFC back to printable status is to:

- edit AVC.var.tool to set `lane_loaded`
- edit AFC.var.unit to set `tool_loaded: true` on the correct leg

Ideally there should be a command like SET_LANE_LOADED LANE=leg2 which would set these variables, allowing a `TOOL_UNLOAD` or `Tx` command to work correctly with no file editing.

jimmyjon711, 2024-12-27:
Thanks for requesting a feature to be added, we already have something for this in our internal tracking of things to add

jimmyjon711, 2025-02-17:
This has been Added in #256

closed 2025-02-23 by MG-longshot