AFCProject/AFC-Klipper-Add-On #366: Adding retry logic when loading lanes and failures happen

ejsears, 2025-05-10:
<https://discord.com/channels/1229586267671629945/1341187458380992512/1341459453647781908>

dohlin, 2025-06-03:
I would LOVE LOVE LOVE to see this added as an option QOL feature - definitely needed for those rare edge cases where the filament just happens to enter at just the wrong angle or a slightly rough edge of a cut gets caught in the extruder entrance.

A way to specify how long AFC will attempt to unload from toolhead before throwing an error would also be really nice - I have noticed my Orbiter 2 filament sensor works great, but it doesn't trigger as instantaneously as I'd wish, which creates some edge cases where unloads can believe they have failed even though the filament is out of the gears already - just because the sensor was a second or whatever delayed in signaling it was unloaded.

But definitely 110% would love to see this reload/retry logic with a customizable variable to allow for, say, ~3 attempts.

mikepiko, 2025-06-14:
Would be nice to have the Toolhead "wiggle" a bit when one of the Switches in the Toolhead report that they are still pressed. Have 1 in 10 times of filament change that the upper switch said "pressed" when it cleary needed one little wiggle to move the ball bearing.