AFCProject/AFC-Klipper-Add-On #408: [Enhancement]:  Enable auto calibration of extruder rotation parameter

rsarwar87, 2025-05-20:
### What are you looking to see?

Enable auto calibration of extruder rotation parameter.

### Why do you want this feature?

0. set new param on AFC_buffer: multiplier_default
1. Reset multiplier_default to one at the start of calib
2. extrude 100 mm
3. calculate new multiplier_default
4. save

rsarwar87, 2025-05-20:
I was thinking of making something similar. Does anyone have any suggestions? We could just modify the rotation_distance as well, I guess.

ejsears, 2025-05-20:
@rsarwar87 Personally, I've never adjusted the `rotation_distance` on my BTs at all, I've just always set it to what it is, and then it was usually close enough that it just worked, or the TN adjusted it for me. 

However, I'm personally all for automation and making things easier. 

I can't speak to the code part, I'm not nearly as well-versed on the python side of this stuff.

jimmyjon711, 2025-05-20:
@rsarwar87 I think I am failing to see what this is trying to fix, and currently I don't think there is a way to know that the filament moved exactly 100mm. in my opinion leaving this as a manual calculation for rotation distance by the user that gets it close enough and then just let the TN buffer do it's job to take care of the difference by changing the rotation distance while printing with the multipliers should be fine.

rsarwar87, 2025-05-20:
We can ask it to move 100 mm, and then use a calliper, so yes, it is a two-step process and not fully automated without an encoder.

rsarwar87, 2025-05-20:
EDIT: actually, if we have two sensors in the toolhead, then it is possible to do the automation with the gear latch disengaged.

rsarwar87, 2025-05-20:
yes, that is what i am suggesting. I made two print, one at 7 pm the other at 11 pm on May 19th. this screenshot is the history of the status of the Expanded switch on the TN on my Home Assistant. grey is clear, yellow is triggered. and during the prints, the switch was triggered and never cleared.

rsarwar87, 2025-05-21:
> > yes, that is what i am suggesting. I made two print, one at 7 pm the other at 11 pm on May 19th. this screenshot is the history of the status of the Expanded switch on the TN on my Home Assistant. grey is clear, yellow is triggered. and during the prints yesterday, the expanded switch was triggered and never cleared.
> 
> Ok I think I'm following now. So your envision for this issue would be a semi-auto calibration step to walk a user through calibrating their current rotation per lane, and then just updating their rotation distance in their config file so that it better matches their motors for those that are not using BOM motors/different motors like in your QB?

Yes, I think the problem comes from changes made by update_rotation_distance  not being presented, which makes the manual calibration of the rotation distance erroneous. Therefore, a dedicated macro to handle this would be ideal in my opinion. 

currently I having to use a rather large value for the multiplier_low of 0.85 , https://github.com/rsarwar87/voron2.4_250_settings/blob/master/AFC/AFC_QuattroBox.cfg#L12