AFCProject/AFC-Klipper-Add-On #425: Cut macro reversing servo movement

Planet1994, 2025-06-05:
Line 109
> When running cut macro the servo would go in reverse direction when commanded. Noticed that this line should read `_CUTTER_SERVO POS=in` 

Additionally;
Line 195 should read
>``_CUTTER_SERVO POS=out`

 _Originally posted by @Planet1994 in [b448e6c](https://github.com/ArmoredTurtle/AFC-Klipper-Add-On/commit/b448e6c24325785082813b98c673b7d807fe28d6#r159126091)_

Planet1994, 2025-06-05:
I cannot check that at the moment, though my question would be how would in/out be defined? I was reading "in" as in the print area and "out" as out of the print area.

jimmyjon711, 2025-06-05:
> I cannot check that at the moment, though my question would be how would in/out be defined? I was reading "in" as in the print area and "out" as out of the print area.

In/out is more for the movement of the server when cutting, so move `out` to do the cut and move `in` to retract the blade back

Planet1994, 2025-06-05:
I see, so think more like a pocket knife. But like you said, it's more of a template and as long as my use case/modifications work then there are no issues.

closed 2025-06-05 by Planet1994